rkaczorek/astroberry-server #190: No filterwheel for SBIG in latest build

jojje70, 2022-08-12:
I cant get the filterwheel to initiate in the latest build of Astroberry. No error messages. I went back to the build from 29/1-22 and everything works fine. 

Attached is a CCD+filterwheel-log from the latest build and two INDI Control Panel Messages, one from the latest build and one from the Januari build (if it is any help).

/Jojje
[logs.zip](https://github.com/rkaczorek/astroberry-server/files/9314990/logs.zip)

rkaczorek, 2022-08-15:
This driver is not maintained by me so I cannot do much about it. I have seen your post on [INDI forum](https://indilib.org/forum/general/12138-sbig-driver-no-longer-recognizes-cw8-filterwheel-fuer-sbig-stf8300m.html#85283) on this topic. I would suggest you report it also to the original INDI driver's [source repository](https://github.com/indilib/indi-3rdparty).

jojje70, 2022-08-15:
Yes, I saw the post on the INDI forum as well after I posted on GitHub, but that wasn´t from me, just another user of the SBIG-8300 camera. Didn´t you respond that you have posted a fix for this?

I will try to update, but if the filterwheel is still not initiate I will report it to the original drivers source repository as you suggest.

Thanks for your response.

/Georg

Skickades från E-post<https://go.microsoft.com/fwlink/?LinkId=550986> för Windows

jojje70, 2022-08-16:
After a recompile I can happily report that the filterwheel is now initiating again.

jojje70, 2022-08-16:
Closing this issue, see comments.

closed 2022-08-16 by jojje70

rkaczorek, 2022-08-16:
I have just pushed fixed version to astroberry repository

jojje70, 2022-10-11:
I made a test upgrading from the online software repository as well, and everything worked like a charm, including the filterwheel. Many thanks!

Skickades från E-post<https://go.microsoft.com/fwlink/?LinkId=550986> för Windows