rkaczorek/astroberry-server #193: The previously saved values in the INDI Control Panel are no longer available!

AlbertFetscher, 2022-08-25:
Raspberry Pi 4
Astroberry (updated 25.08.2022)
Connected devices:
	USB3.0 SanDisk Ultra 32GB memory stick
	USB2.0 iOptron GEM28
	USB2.0 iOptron iGuider
Connected via VNC client over WLAN

Test steps:
1) Start KStars
2) KStars -> button "Toggle EKOS"
3) EKOS -> button "Start"
4) INDI Control Panel, tab "V4L2 CCD":
	a) Note: Options previous set are not available! 
	b) Set some options again and saved it:
	b1) "Capture Options": set "Capture Size" to 1280x960
	b2) "Image Info": set "Max. Width", "Max. Height", "Pixel size (um)", "Pixel size X", "Pixel soze Y" to the correct values: 1280, 960, 3.75, 3.75, 3.75 and then click button "Set" (of course)
	b3) "Image Settings": "Frame" indicates red, therefore need to click button "Set"
	b4) "Options": "Configuration" click button "Save" to save configuratoin (of course)
5) EKOS -> "Disconnect" 
6) Close window "EKOS"
7) Close window "KStars"
8) Reboot Astroberry... and wait until VNC client is ready to reconnect
9) Repeat test steps 1, 2 and 3
10) INDI Control Panel, tab "V4L2 CCD":
	a) Check if all previous set options (test step 4) are available

Expected result:
	Previous set options are available.

Result:
	Options previous saved are not available!
	
Issue:
	Test step 10a, in a previous session I set the correct options, saved it but now these values are not available. Why?
	
(If you need screenshots, don't hesitate to ask.)

rkaczorek, 2022-08-25:
Check the content of /home/astroberry/.indi directory. This is where all drivers' configuration is saved. Major changes to INDI and Kstars have been introduced in the last update. I noticed that some configs changed their names. If you find configs (xml files) similarly named but with different last updated dates you are probably facing the same issue. In my case saving new configuration works ok. Also make sure that in KStars Settings / EKos you enabled Load device configuration at On connection. Finally try to manually Load config (in INDI panel) after connecting a driver. Let me know how it goes.

BTW. These type of issues i.e. related to KStars and INDI should be reported at upstream repositories:
- INDI https://github.com/indilib
- Kstars https://invent.kde.org/education/kstars/-/issues

AlbertFetscher, 2022-08-26:
Thanks for your reply!

"Check the content of /home/astroberry/.indi directory. This is where all drivers' configuration is saved. Major changes to INDI and Kstars have been introduced in the last update. I noticed that some configs changed their names. If you find configs (xml files) similarly named but with different last updated dates you are probably facing the same issue. In my case saving new configuration works ok. "

=> Checked and it is OK.

=> But missing some entries in XML file "V4L2 CCD_config.XML" related to tab "CCD_INFO"! 
   Key/Value for: CCD_MAX_X, CCD_MAX_Y, CCD_PIXEL_SIZE, CCD_PIXEL_SIZE_X, CCD_PIXEL_SIZE_Y, CCD_BITSPERPIXEL
   
=> But missing also some entries in XML file "V4L2 CCD_config.XML" related to tab "Capture Options"! 
   Key/Value for: e.g. "Capture Size"

"Also make sure that in KStars Settings / EKos you enabled Load device configuration at On connection. "

=> Checked and it is OK.

"Finally try to manually Load config (in INDI panel) after connecting a driver. Let me know how it goes."

=> Checked and it is OK.

=> But when saving the options or changing a value, my previously manually changed entries in the XML file are ignored and are then missing from the XML file. This information is now lost.