rkaczorek/astroberry-server #194: Bullseye support

juandanielmorcilloregueiro, 2022-09-01:
Good afternoon,

I had to compile KStar from code due an issue in the last release. However, to compile is mandatory use [Bullseye version of Raspbian](https://indilib.org/forum/development/12093-cmake-failing-not-finding-a-configuration-file-for-package-ecm.html), so I´m trying to do a fresh instalation of Bullseye only to compile KStar and after that install it in Astroberry.

Do you plan to release a new version of Astroberry with Bullseye? I think it is very needed for this kind of situations.

Thanks!

rkaczorek, 2022-09-01:
As the matter of fact I have just pushed bullseye arm64 packages to the repo. You can add the repo to fresh install of Raspberry OS 64bit by running:
```
wget -O - https://www.astroberry.io/repo/key | sudo apt-key add -
sudo su -c "echo 'deb https://www.astroberry.io/repo/ bullseye main' > /etc/apt/sources.list.d/astroberry.list"
```
Then run `sudo apt install indi-full kstars-bleeding phd2` to get core apps installed

PS. Don't tell anyone yet ;-)

rkaczorek, 2022-09-01:
BTW. astroberry-server-full is not ready for bullseye yet

juandanielmorcilloregueiro, 2022-09-01:
Thanks you for the quick response! Yes, I faced when I tried to installl server-full :P

With the default KStar installation I got an error to launch. Right now I am compiling with my RP3. I hope that version works fine.

Best regards.

nigeldun, 2022-11-18:
Is there any update on the progress to 64bit?