rkaczorek/astroberry-server #195: ASI585MC not working

Tmackos, 2022-09-02:
Running astroberry on a Pi3B+ latest version 2.0.4 and PC with latest Ekos/Kstars 3.6.0. The camera is connected to the Pi which is connected to my network via WiFi. I'm using the PC to remotely control the Astroberry.
My ASI585MC is not recognized in ekos, simply doesn't show up. In ekos I have tried to configure the camera as ZWO CCD and ZWO Camera 1. After initializing the profile ASI585MC doesn't show up.
ASI585MC is relativly new camera will there be support for this camera in the future in a furture SW release?
ASI290MM works fine on the setup, as well as my Nikon D7000 DSLR.

Tmackos, 2022-09-02:
[image]

rkaczorek, 2022-09-02:
It is probably not supported (yet) by ZWO libraries used in INDI driver. They are frequently updated so stay tuned

Tmackos, 2022-09-03:
Sounds good!
To update the indi drivers do I need to update the Astroberry or Kstars/Ekos? Thanks.

rkaczorek, 2022-09-03:
INDI drivers only. Just a regular `sudo apt update && sudo apt upgrade`

TonyF1330, 2022-09-12:
I also recently swapped from a DSLR to an ASI585MC. I've seen some odd behavior as well, but I am able to connect to the 585 and use it to image in EKOS. I believe I'm also on Astroberry 2.04, but I'm running EKOS directly/locally on the pi and remoting in with VNC while I think Thomas is connecting to the INDI server on the Pi from KStars/EKOS on Windows? Maybe it works one way and not the other right now?

[image]

How do I determine which version of Astroberry I'm running so I can confirm that it's 2.04? (I'm a linux newbie still)

Here's a quick shot of the Ring Nebula back on 8/29/2022 with the 585 on Astroberry/EKOS: 
[image]

I have noticed that I'm unable to use the Live Video in EKOS with the 585. EKOS connects and displays a very slowly updating video, but the video is unusably slow and the Pi's CPU pegs at near 100% until I turn the Live video off. I assume this is something that will get better as better drivers are available. But single image captures have been working well enough for me.

I hope this provides another useful data point in your investigation.

vidrazor, 2022-09-13:
Hmm, I guess the issue is the 585 chip, or is it the API? I've considered getting the Player One Uranus-C camera based on the same chip. A look in the profile setup only shows a generic Player One driver. Unless they all run on a global API, I suspect Ekos will not recognize the new Player One Uranus-C camera as well.

rkaczorek, 2022-09-21:
> How do I determine which version of Astroberry I'm running

`cat /etc/astroberry.version`

TonyF1330, 2022-09-21:
I'm on 2.0.5, not 2.0.4 as I incorrectly indicated in my earlier message.

Tmackos, 2022-10-04:
Reports here that ASI585MC works on Stellarmate.
https://indilib.org/forum/ccds-dslrs/12006-asi585mc-is-this-supported-by-latest-indi-asi-driver-possiblyekos.html

Stellarmate have updated Indidrivers based on this link:
https://www.stellarmate.com/products/stellarmate-os.html
https://github.com/indilib/indi/releases

Astroberry 2.0.4 must be left at indi driver version 1.9.0 ish

Any release plans to include new indi drivers to a new Astroberry release anytime soon?

vidrazor, 2022-10-04:
How do you find out what INDI version you're running? I'm listing Astroberry 2.0.4 and KStars 3.5.8.

Tmackos, 2022-10-04:
> I have the same issues with a new ZWO ASI533 in Ubuntu OS after a recent KSTARS/EKOS 3.6.1 from 3.5.(?) upgrade, the 533 is recognized but KSTARS/EKOS crash immediately when an exposure is executed. Now the ZWO ASI178 crashes too. Rebooted many times and tried new profiles with 1 camera only, same problem. I didn’t try the ASI 120 guide camera. The Nikon D5600 DSLR continues to work fine. Fortunately the PI-4s have not been upgraded in many months (KSTARS 3.5.7? and old INDI drivers) and the 533 seems to work fine. Bill
> […](#)
> Sent from my iPad

Same issue? Different camera and different problem.

closed 2026-02-11 by rkaczorek