rkaczorek/astroberry-server #201: allow other vnc servers otehrn the realVnc with astroberry-server-wui

uriba107, 2022-10-18:
Currently `astroberry-server-wui` requires realvnc-server package to be installed.
normally it's not a real issue, as astroberry is intended to be installed on piOS.

however, I'm using the astroberry repo to install ona clean Debian on a different ARMv7 32bit SoC. (IMx6). for the most part everything works, but it will not allow me to install the package as realVNC is only providing armhf packages for Rpi hardware. 

I've configured x11vnc on my little SoC and it works, but the dependency on realVNC is a problem. if it would be possible to allow realvnc-server OR x11vnc (or any other vnc server available for Debian like tigervnc) it would be great!

no need to actually configure anything, let the next Idiot that tries pulling off this type of shenanigans figure it out by him/her self.

but with the unavailablity of Raspberry pies, allowing people to hack their way to glory is great fun :)

Thanks for all the hard work!
Uri

a proposed fix will be to put something like this in the package `control` file
```
Depends: libindi1 (>= 1.7), libindi-data (>= 1.7), indi-bin (>= 1.7), nginx, websockify, tightvncserver | tigervnc-standalone-server | realvnc-vnc-server, python3-pip, python3-bottle, python3-flask, python3-flask-socketio, python3-gevent, python3-numpy, python3-pil, gpsd
```

rkaczorek, 2022-10-18:
Thanks for reporting. Good point. It will be fixed in the next release.

SteelersForever, 2023-01-09:
Hey @uriba107 can I talk to you?

https://www.digitalcombatsimulator.com/en/files/3327489/

This file contains the Tu-204/214 Passenger/ELINT aircraft. Would you be willing to add that to your Hound ELINT?

uriba107, 2023-01-09:
> Hey @uriba107 can I talk to you?
> 
> https://www.digitalcombatsimulator.com/en/files/3327489/
> 
> This file contains the Tu-204/214 Passenger/ELINT aircraft. Would you be willing to add that to your Hound ELINT?

Sorry if it's going to sound agressive.
But it's quite rude to comment on a project issue with something related to. A completely different project.

Hound is it's own project. Please open an issue there.

SteelersForever, 2023-01-10:
@uriba107 I am so, so, so sorry for this.

But I am new and have no clue how to open the issue on Hound's repo.

closed 2026-02-11 by rkaczorek