rkaczorek/astroberry-server #202: Firecapture 2.6 to 2.7 drops FPS 25fps to 4fps

Psychorages, 2022-10-25:
Hello,

First i guess i should post this issue to Firecapture  ... but i'd like to share the experience there :)
(so i won't mind if someone decide to reject this ^^ )

- I install Astroberry on my Pi4 (on a SSD). 
- I start firecapture => my Zwo 178 MC give 25ps ( full resolution, no debayer, 8bit, 40ms, HighSpeed, usb70%)
- I copy the content of /opt/firecapture to /opt/firecapture2608 for futur comparison
- I download and install Firecapture2.7 (the same with or full apt-update, upgrade ...)
- I start firecapture 2.7 => my Zwo 178 MC give 4fps ( same cfg :  full resolution, no debayer, 8bit, 40ms, HighSpeed, usb70%)
- I start again firecapture 2.6 => my Zwo 178 MC give 25fps ( same cfg :  full resolution, no debayer, 8bit, 40ms, HighSpeed, usb70%)

I tried to use librairies ".so" coming from 2.6 in 2.7. No change

rkaczorek, 2022-10-26:
I cannot comment on this as astroberry only repackages this app as it comes from firecapture developer.
If you find any reasonable answer after contacting firecapture developer, please share it here as well.

Psychorages, 2022-10-26:
> I cannot comment on this as astroberry only repackages this app as it comes from firecapture developer. If you find any reasonable answer after contacting firecapture developer, please share it here as well.

That's my goal :)
I effectively have sent the same post in Firecapture contact form.

Thx.