rkaczorek/astroberry-server #204: Unable to connect using browser

nigeldun, 2022-11-18:
Previously have been able to connect using browser but now seem unable to. This has started occurring with a windows pc and an ipad. VNC is still working fine. I can also use the web interface to start the indi server it’s just the connection to the desktop that is failing.

bbillp, 2022-11-19:
Maybe Astroberry went in to hotspot mode and you didn’t notice it.   Check the top right corner of the Pi screen.Sent from my iPad

nigeldun, 2022-11-19:
My PC was connecting to the astroberry via a wifi router which afaik only allows the wifi dongle I’ve added to the astroberry to connect. Also I could still use the icons on the left side to setup and start indi server and I was also able to use kstars from my pc to control it. The same was the case with the ipad though that can connect to the hotspot directly but I’m pretty sure I was too far away.

I’m having a couple of problems since I applied updates that came up over the last couple of weeks. My wifi router has also been applying auto updates so there’s a few moving parts. I will see if it works via the hotspot in case it’s the router.

nigeldun, 2022-11-24:
Turns out I cannot even connect to the novnc demo site so it’s not down to astroberry. My fritzbox router has been getting a lot of updates recently so maybe it’s something to do with that.

Closing issue thanks

closed 2022-11-24 by nigeldun