rkaczorek/astroberry-server #206: Update Astroberry Repositories ?

bbillp, 2023-01-12:
Thanks for Astroberry, I am a long time user but when I try to use Astroberry with Kstars 3.6.0 / EKOS / INDI from the night of January 5, 2023 with a new IMAGE from scratch for the ZWO AM5 mount.   I’ve found problems in the EKOS panel DATE/UTC.  Apparently the  system fails to communicate with the AM5 to Get or Update the AM5 Date/Time/UTC so the AM5 mount will not go into EQ mode (RED LED) and just sites idle.   The Log error message is ‘Can’t set UTC’.

Is it possible to update to update the repositories with the very latest versions of each?  

For now The ‘get around’ is to start the AM5 mount using the ZWO AM5 APP to synchronize this information then simply start Astroberry and continue.    

thank you …

rkaczorek, 2024-09-13:
A new major version will be released this year. It is built from scratch based on Debian 12 (bookworm) and latest INDI, 3rd-party drivers, Kstars, PHD2 and much more. I'm aware that it has taken ages since  the last update to previous version. However breaking changes we introduced to KStars some time ago, which made impossible to update the system based on Debian 10 (buster). The entire system needed to be rebuilt and the new packaging environment had to be set up. Also new front-end is being developed. I'm hoping to find some time and release the new Astroberry OS soon.

jochym, 2024-09-13:
Great news @rkaczorek . I was seriously afraid that you have abandoned the project. It must be a huge undertaking. Rest assured that it is really appreciated. If I can be of any help, let me know. I do not have much free time, but I will gladly offer the little I have to this project.

guilleortas, 2024-09-13:
That's great to hear! Your efforts are really appreciated. Please count on the community for testing or any other help we can offer

PeteCad, 2024-09-21:
Thank you very much for keeping with the project.

cgobat, 2025-11-14:
Any update on the status of this, @rkaczorek?

rkaczorek, 2025-11-16:
Should be published by the end of the year. Stay tuned!

froeschel2001, 2025-11-16:
Thank you!
Will the new released version be compitable with the Pi 5?

cgobat, 2025-11-16:
That's great to hear—looking forward to it!

cgobat, 2026-01-05:
Any update on this timeline, @rkaczorek?