rkaczorek/astroberry-server #208: USB Type C - failure to detect ASI120mm camera

josephrenda, 2023-01-17:
I have a new ASI120mm guide scope which works perfectly fine on windows 10/11 with native or ASCOM drivers; 

But not on the latest astroberry release. 

The Type C to USB A plug (which is good, and I tried 2 different ones which also work in windows) does not seem to enumerate on lsusb 

See dmesg error. 

[image]

paulborgermans, 2023-01-18:
Did you update the astroberry to the latest version (including the updates for the upstream raspberry os)?

josephrenda, 2023-01-18:
Problem resolved.  The original zwo (30cm) cable was damaged in a way that did not function on any raspberry pi OS. (Debian, ubuntu, or Raspbian)  but worked on a PC with both ubuntu and windows 10/11.  Changing the cable resolved the issue;    From: Paul BorgermansSent: Wednesday, January 18, 2023 8:55 AMTo: rkaczorek/astroberry-serverCc: josephrenda; AuthorSubject: Re: [rkaczorek/astroberry-server] USB Type C - failure to detect ASI120mm camera (Issue #208) Did you update the astroberry to the latest version (including the updates for the upstream raspberry os)?

closed 2026-02-11 by rkaczorek