rkaczorek/astroberry-server #209: Astroberry Pi-4 Schedular Help  Needed, Kstars 3.6.0 1/5/2023 updates

bbillp, 2023-01-19:
[image]
[image]
[image]
[image]

Maybe I am overlooking something in the scheduler because it does not activate the ZWO AM5 Mount.  I can do individual GoTos and create lines in the Schedular and start the schedular successfully but then the scheduler doesn’t process the line item and eventually aborts the line and all the following lines.    

Is there something I’m overlooking ?

jnowat, 2023-01-20:
Sir I see you have impeccable taste in laptops. Writing you from a T420 on
her 10th year with me. Jealous of your surviving red button.

Have you tried using the unpark mount feature for the jobs? See if you get
a message like "Mount is already unparked". I check that box out of habit,
and that particular message brings me some amount of solace as the schedule
begins.

bbillp, 2023-01-20:
Thank you for the reply I think I’ve tried so many combinations that I have to say yes I’ve checked in and checked the Hunt Park with no change. When using Ubuntu 20.2 The first Line in the scheduler will go to but fails to execute the camera fails to go to the next line.

Go to in kdtars works and the camera tab works. The problem is strictly in the scheduler tab.

The Lenovo ThinkPad T430 ass seems like the perfect machine I plan on putting in another hard drive and dual boot windows 10 and make a Linux partition then try to sing my present 20.2 Ubuntu do the Linux partition on the new drive. I don’t know if that’s possible. Ubuntu 22.04 doesn’t have K stars or Indy in the software load folders. So I’m stuck.

jnowat, 2023-01-21:
I've had problems with some go-tos as well. They were all in the west (pier
side east) when I was currently pier side west. I had to disconnect/connect
the mount a few times and even go outside to the mount and manually slee to
puer side east. That was with a Losmandy Gemini, not an ASI machine.

Can it be either side of the meridian? Kay be some clues to positional
effects. I couldn't go-to if it required a meridian flip, is what I'm
saying.

Did the same exact thing with my thinkpad. They don't make rn like they
used to- I even upgraded the RAM because it had an empty DIMM slot!!

For Ubuntu, have you tried: sudo apt-get install indi-full kstars-bleeding
gsc

bbillp, 2023-01-21:
Ubuntu 20.04 on a laptop, ZWO AM5 mount, ZWO cameras, KSTARS 3.6.2 or 3
Similar schedule problems Astroberry image 1/5/2023 
I added ‘unpark’ and ‘park’ to the scheduler and here is what I experienced.
-Single line scheduler entry works only if it is the first attempt.
-Multiple lines in the scheduler entry work ONLY if there are NO Previous errors
-ANY type of error in the scheduler requires a COMPLETE Power off/on cycle to clear everything as just restarting KSTARS/EKOS does not clear previous entries.
-If I have successful multiline Schedule It WILL NOT RUN A SECOND TIME as EKOS simple aborts each line.  A full Power cycle is necessary .
POSITIVE NOTE: I use the scheduler more and more often even for single line entries and while experimenting with the Mosaic feature, pretty cool !!!!

Otherwise I really like KSTARS/EKOS and Astroberry and hope the Developers continue their work.

bbillp, 2023-01-21:
PS: I am now afraid to ‘update/upgrade’ Ubuntu because upgrading might update 20.04 32 bit to version 22.xx 64 bit version and there doesn’t seem be be 64 bit KSTARS/EKOS INDI available. I tried finding KSTARS/EKOS INDI 64 bit on a separate Ubuntu 22 machine …..  this software does not exist.

jnowat, 2023-01-21:
I'm running into similar problems with my Astroberry losing connection to
the server. Cycling power seems to be necessary to reconnect. I shave not
yet, but I suggest watching the logs video on indi website and submitting
logs here and/or there. Go ahead and update/upgrade, but *after* you've
backed up the SD card (or file system) that has been working best for you.
I learned this in such a hard way a couple years ago.

Sorry I couldn't be more help, but like you I've been taking the good with
the bad. I also love the mosaic tool by the way- I do wish, however, that I
could turn off rotation!

bbillp, 2023-01-22:
> That was with a Losmandy Gemini, not an ASI machine.
> […](#)

The Losmandy Gemini 2 is weird with or without a PC.  I returned my NEW GM811 because one of the terrible problems was random slewing out of control.   I do recall for best computer GoTo to perform a hand controller goto in the same area of the sky then initiate the PC software.

bbillp, 2023-01-22:
I improved my connection by deleting the local connection and only use the hotspot connection which I changed to the 5ghz wifi band.  I’ve 2 Raspberry Pi-4 dedicated to separate mounts and this connectivity has been solid.    Love the mosaic tool and really don’t need the scheduler since I put my gear back into the house at the end of the night.    I plan to make the lens cap cover project mentioned someplace here.   Sent from my iPad

closed 2026-02-11 by rkaczorek