rkaczorek/astroberry-server #214: Bulb mode does not work with the Nikon Driver - Using a Nikon D200

antonical, 2023-02-17:
Hello I hope someone can assist. I am struggling to get EKOS and INDI working with my Nikon D200. The driver seems to work ok when Force bulb mode is off upto 30 secs. This is when bulb mode is already selected on the camera which is limited to 30-second exposures from the actual camera settings.

If you set force bulb mode on and try and capture any image it fails.The driver reports [INFO] Force BULB is enabled. All exposures shall be captured in BULB mode except for subsecond captures. But they are not. The shutter is not triggered and the camera display just shows the PC with the moving line until you switch it off and it then messes up the camera settings. The quality setting is changed etc.

So you have to disconnect in the INDI control panel, Stop EKOS switch off the camera and start again.

So if I want to take any exposure over 30 seconds it does not work.

My intervalometer works for longer exposures but of course I have to be at the camera for that to work!

Can anyone help?

Cheers
Spart

antonical, 2023-02-19:
With gphot2 using entangle to control the camera. Then you are limited to 30 seconds exposure and this works reliably and consistently upto 30 sec exposures. However setting bulb mode on the camera and then connecting results in it not functioning. Gphot2 sets PC Mode on the camera LCD and is the same way that Nikon Capture software does. However, when connected via INDI the LCD camera screen is still operational. I don't know if this is how it is supposed ot work but all documentation says when connected via USB then it should show PC mode in the LCD.

Also, the Quality setting is not quite right. When you set the quality to RAW in the camera and connect it then it is displayed as unknown value 003 rather than RAW. Images captured seem to still be in RAW .NEF format. I think it is actually setting RAW with Lossless Compression. Rather than just RAW with no compression. Changing the setting in the software to RAW results in it setting RAW plus jpeg. This is a serious PITA as it fills up the SD card with useless jpg files. There is a size difference in the 2 .NEF files with the RAW software setting resulting in the larger file so probably RAW no compression but also a jpg. RAW setting on the camera then connect you get the 003 error but lossless compressed .nef and no jpg.

Any help appreciated.
Cheers

nombreuser, 2023-02-27:
Hello,

Please refer to the "Bulb" section in this page:
http://www.gphoto.org/doc/remote/

To summarize: only newer Nikon models support correct bulb capture.
Please also look at this page: http://www.gphoto.org/proj/libgphoto2/support.php
I'd say that mostly every Nikon which uses non-PTP drivers would support it, otherwise you're stuck with 30 seconds.

Thats all I found on the internet while trying to fix the same issue with my old Nikon.