rkaczorek/astroberry-server #218: Astroberry Update Renders Wifi Inop

costargazer, 2023-04-03:
I have tried this procedure twice to verify this is what is happening.

1st boot of Astroberry from SD card, run update then upgrade.  Changed the display resolution and reboot after upgrade.  Wifi will not connect to router/modem, unable to make direct connection with wifi hotspot.

Reload Astroberry using Raspberry Pi Imager on SD card, 1st boot with fresh Astroberry and wifi is back to normal.

bbillp, 2023-04-03:
Left or right Click the WiFi icon in the upper right corner to see if the Local WiFi is on or the Hotspot is on.   Normally when Astroberry sees WiFi signals it will not turn on the hotspot but by clicking the properties of the WiFi icon you can disable the wifi address which will then keep the HOTSPOT on.   You might have to plug in the CAT5 cable to make adjustments.     Sent from my iPad

costargazer, 2023-04-04:
I should have given more information.  The Rpi wifi will try to connect to the router but times out and the two computer icon appears.  Checking network settings and Rpi settings everything appears normal but the Rpi onboard wifi appears disabled.
Wifi hotspot is disabled also when I switch to that.

Also noticed with a fresh 1st boot and changing the screen resolution to 1024 X 768 60hz the wifi is disabled also even without updating the software.  Change the resolution back to 1240 X 800 and the wifi is back to normal.
The only fix I know of is to do a fresh reload of Astroberry on the SD card.  I haven't updated the software and I haven't had any  problem since.

bbillp, 2023-04-05:
I am curious of your HDMI cable is the culprit by emitting high level RF interfering with the WiFi signal because I just experienced this on my laptop with a bluetooth dongle up against the HDMI cable.  Try going headless with vnc vewer on the host PC to the hotspot.Sent from my iPad

costargazer, 2023-04-05:
First of all, my Rpi wifi and direct connect is working normal now because I reflashed the SD card and booted with that.  

If I don't do an update everything is working.  Bill it is not the HDMI cable, otherwise it would happen all the time.

My intention for posting this was to inform the developer that there is a glitch in the update that causes this.

As long as I don't update I'm working fine.

closed 2026-02-11 by rkaczorek