rkaczorek/astroberry-server #222: Certificate error on the website

N7Normandy, 2023-08-02:
Hello everyone. There is a certificate error on the website that expired on 08/31/23. As a result, the repo can no longer be updated. Untrustworthy. Is the repo no longer available? Thanks in advance :)

Error:
Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 46.105.204.23 443]

CreasaBE, 2023-08-02:
Yep same issue here when trying to set up indi-allsky as it needs packages from astroberry. Cert has indeed expired and needs to be renewed again.
It's a LetsEncrypt certificate, so it should've been done automaticaly, but apparently it has not.

N7Normandy, 2023-08-02:
Yeah it works . Thanks a lot.

rkaczorek, 2023-08-04:
HI guys. This is error on the server side. For some unknown reason SSL certificates did not renewed recently.
I'm gonna manually fix this right away.

guilleortas, 2024-08-30:
It's still happening for me on a fresh install downloaded and flashed today

rkaczorek, 2024-09-02:
Please remember that Raspberry Pi is not (by default) equipped with real time clock, which keeps system time between reboots. It gets current time from the Internet and it takes some time, especially in case of fresh install.
Now, if you try to run update too quickly after system boot (and your system time has not been updated yet ), you will always get certificate error on any software repository. It is a system feature, not a bug.
If it happens, you can check your time running `date` command in terminal window, wait a few seconds, run the command again until it shows proper date. Only then you can run system updates.