rkaczorek/astroberry-server #224: Repository no longer has a Release file

lastxmas, 2023-08-04:
I'm using an RP4 and am now getting an error on top of the certificate not trusted error that says astroberry[dot]io/repo buster Release no longer has a Release file. (paraphrasing the actual error)

jmvm-nz, 2023-08-05:
Yeah, same issue for me, it appears the certificate issued to the website is using the wrong name, as you can see if you try manually browse to the site https://www.astroberry.io/repo/
That cert was issued for DNS Name: hosting.cdn.ovh.net and DNS Name: www.hosting.cdn.ovh.net not for astroberry.io

lastxmas, 2023-08-05:
I was able to bypass the wrong-name issue but the no release file is something I haven't figured out how to get around yet. I'm trying to get the new indi files to cover the SVBony SV605CC camera (only SV305 is showing up in my CCD list and it doesn't recognize the camera).

rkaczorek, 2023-08-06:
Certificate issue should be fixed shortly.
I'm not sure why would you have problems with accessing [Release file](https://www.astroberry.io/repo/dists/buster/Release)

closed 2026-02-11 by rkaczorek