rkaczorek/astroberry-server #225: Raspberrypi 5 support

Fernien, 2023-12-07:
Hey there,

is there an update planned to support Raspberrypi 5 with the newest OS?

Best,
Niklas

brianx87, 2024-05-19:
Is there any update upon this? I thought it was expected in December

rkaczorek, 2024-05-19:
Working on it. Originally planned to be released end of December 2023. However significant technological changes both Raspberry Pi 5 and Raspberry OS resulted in postponing the release. This is going to be a major release for upcoming years and the goal is to build stable and long term support system that includes as much latest technology and solutions as possible.

LeftMan24, 2024-07-04:
Any news about the update for pi 5

nystire, 2024-07-30:
Is there anything that people can do to help with this? Testing, perhaps?

rkaczorek, 2024-07-30:
It's going to be released later this year. Stay tuned

hidrab70, 2024-08-08:
I understand that you are thinking that it is for RPI5 when most users have RPI4. Will it work with RPI4?

Regards

codykingham, 2024-09-24:
Eagerly awaiting this release!

rkaczorek, 2024-09-26:
@hidrab70 Will run on RPi4 too

hidrab70, 2024-09-27:
Muchas gracias, ya estoy deseando verlo.

froeschel2001, 2024-10-03:
Is the release of the RPi5 version still planed for this year?

jayjar291, 2024-12-11:
Bump Really looking forward to this especially with the recent release of the CM5