rkaczorek/astroberry-server #228: I lose wifi hotspot when camera connected to usb3.

costargazer, 2024-03-29:
Interesting problem.  I have two raspberry pi 4b, one with 8gb, one with 4gb.  I am able to log onto the wifi hotspot remotely, then when I plug my camera into the usb3 on the 8gb model I lose the wifi hotspot.  It doesn't happen with the 4gb model.  I flashed the micro sd card on both with the same img file. Ran the update after the initial install. Searched the issues but did not find anything.
Thank You,
Scott

rkaczorek, 2024-04-02:
It could be well known issue of USB 3.0 and WiFi 2.5GHz conflict coming from h/w design flaw of RPi 4. If the devices connected to USB are exactly the same, I can only guess that these 2 versions of RPi might differ somehow. What you can do is to change hotspot configuration so it uses 5GHz WiFi instead.

costargazer, 2024-04-04:
Yes, I was doing some more research. I'm going to try 5Ghz, and maybe a clamp on ferrite choke on the usb cable.
Thank You

nystire, 2024-07-31:
Did the ferrite choke help?

bbillp, 2024-08-06:
Keep USB cables away from the Raspberry Pi, USB emits strong radio frequency signals to disrupt WiFi and BlueTooth.    Bill Sent from my iPad

closed 2026-02-11 by rkaczorek