rkaczorek/astroberry-server #229: Problem after update the release 2.0.4

MarquezRomulo, 2024-06-11:
Hi there,

I'm having troubles every time I update the OS by **sudo apt update && sudo apt upgrade && sudo apt dist-upgrade** commands. The update finished with error and the consequence the the OS broken after the reboot or power off. Thus I need to install again the OS in the raspi

[image]

First Boot Procedure: https://www.astroberry.io/docs/index.php?title=Astroberry_Server#First_Boot

Some body know how to fix the issue? Apologies but I am not a Linux user so these things are difficult for me.

**OS installed:** Astroberry Server release 2.0.4
**Hardware:** Raspberry Pi 4 Model B
**Micro SD Card:** 64 Gb

Thank you very much!

LightBridge16, 2024-10-20:
Will:
sudo apt-get update
sudo apt-get upgrade
sudo apt dist-upgrade

fit it?

rkaczorek, 2024-10-21:
It is the skychart app that is throwing an error. You can workaround it by removing skychart _before_ updating.
So the command sequence would look like this:
```
sudo remove skychart
sudo apt update
sudo apt upgrade
```