rkaczorek/astroberry-server #231: PlayerOne camera drivers for OaCapture or Firecapture

aday5, 2024-09-05:
Hello everyone.
I am trying to capture with the Oacapture or Firecapture program, but it does not detect my playerOne UranusC camera.
The kstars program does detect it and works, but Oacapture and Firecapture do not recognize the installed INDI-PlayerOne drivers.
Any help? Thank you

rkaczorek, 2024-09-06:
Unfortunatelly Oacapture and Firecapture use their own binary drivers for cameras and other equipment. There is some chance to use system wide drivers while compiling Oacapture but this is quite complex process. You can try to recompile on your own. It's not the case for Firecapture which is distribited as a procompiled binary only.

aday5, 2024-09-06:
I was hoping it would be easier. I have nothing to lose by trying the hard way. Is there somewhere I can follow the instructions to try this? Thanks a lot.

closed 2026-02-11 by rkaczorek