rkaczorek/astroberry-server #232: Kstars not Starting after update & upgrade

schmartdevil, 2024-10-10:
Hello, 

I tried to setup the astroberry as described .
The system works it seems just fine .

But after "sudo apt update && sudo apt upgrade && sudo apt dist-upgrade"
Kstars just don't start anymore.

- kstars Loading screen shows up
- but main window don't 

i there something i could do? 
It seems the 2.0.4 image works. 
but this is also very old.  

thank you for your help

nordcomputer, 2024-10-25:
Just a small side-note: Don`t do a dist-upgrade. The command tries to update to a new Raspberry OS version. Stay on the current version and you should be fine until the new major release drops.

schmartdevil, 2024-10-25:
Ok thank you for this information .
I will try that .

For your information i followed the steps described on the website :
[cid:21ccabdb-4ee3-46d5-9e52-4255571df7d4]

Cheers

Soncebos, 2024-12-24:
Good News.
LG

rkaczorek, 2026-02-11:
Try the new version https://astroberry.io/astroberry-is-back/