rkaczorek/astroberry-server #236: Pi5 / 64bit

egge12, 2024-12-06:
Hi

are there any plans to make Astroberry compatible to Pi5 (bookworm 64bit)? Is there a timeline?

RononDex, 2025-01-07:
The last code push in this repository is literally 4years ago. The project is no longer active and maintained. I highly suggest using a much more up to date system where you can also get newer PHD, Kstars and Ekos versions (the ones provided with astroberry are at best 2-4 years old and can not be upgraded due to debians non-rolling release style).

Personally I use arch linux arm on my RPi5 and it works awesome as my telescope controller (you can see my dotfiles for a bash script that can set it up for you).

jochym, 2025-01-08:
The actual files are not in this repo. So the last push indicates just the last release. Unfortunately, I think that @rkaczorek decided to do the development in-house instead of here. Maybe reconsider @rkaczorek ? I bet there are multiple people who are willing and able to help. I, for one, would do my best. This was a fantastic solution - even for someone who can do similar config for him/her-self. It would be a shame if it got lost in some dark corner of an old hard drive...